pschichtel/JavaCAN #23: Does/Will it work on android?

Deishelon, 2020-10-07:
Hi

Will this work on android? 

Adding
`implementation 'tel.schich:javacan:2.3.0'` to the gradle and trying to use the lib gives this:

` java.lang.LinkageError: Failed to load the native library: /native/libJavaCAN-armv7.so not found.`

pschichtel, 2020-10-09:
I have not tried this on android and I never used JNI on android. If you can provide more information and maybe test a few things I'm sure we can solve this together.

A few questions arise:
1. Does android support embedding resources? JavaCAN detects the system architecture and tries to read it's native library as a resource from the classpath at `/native/libJavaCAN-<arch>.so`, your `<arch>` apparently is armv7, which JavaCAN supports in general.
2. Is the native library included in the APK ?
2. Can I use `System.loadLibrary` from /tmp (or where ever temp files from apps are created on android) ? JavaCAN unpacks the native library to a temp file (using `Files.createTempFile`)
3. Do I need to compile with android specific headers?

pschichtel, 2020-10-11:
Just invested a few minutes looking around, apparently the `System.loadLibrary` should generally work, the problem is probably just the way I unpack the native library.

I think it would be useful to allow providing some kind of library path (maybe even use linux' LD_LIBRARY_PATH to look for the library on standard system paths). That way you could bundle the libraries in the way that works best on android, unpack it yourself and provide JavaCAN with the exact location of where to find it.

pschichtel, 2021-03-08:
I'm closing this issue since there hasn't been any more feedback and with all the changes around library loading since then, especially #31.

closed 2021-03-08 by pschichtel