pschichtel/JavaCAN #24: JVM Crashing

keoir, 2020-11-18:
# JRE version: OpenJDK Runtime Environment (8.0_272-b10) (build 1.8.0_272-b10)
# Java VM: OpenJDK Client VM (25.272-b10 mixed mode linux-aarch32 )
# Problematic frame:
# C  [libJavaCAN.so+0x21c8]

Crashing the JVM when it is using the .so file. On a BeagleBone Black Debian OS.

pschichtel, 2020-11-27:
is it possible to get a full stacktrace?

pschichtel, 2020-11-27:
easiest would be the JVM's crash file, if one exists.

Also SIGILL makes me wonder if it selected the right library for the architecture (ARMv7 from what I've seen) or if something like seccomp might be blocking syscalls here? For the latter I we definitely need more information about where the crash occurred..

psobiech, 2020-12-03:
Hi,
according to this https://github.com/dockcross/dockcross/issues/290 and https://github.com/dockcross/dockcross/tree/master/linux-armv7a the target for beaglebone would be `armv7a` not `armv7`, maybe that is the problem?
 
If you have time, you might try your luck with this: https://github.com/psobiech/JavaCAN/commit/76413e94192ad630e36ec21b46ea571a687d284f (it is not a fix, since it would break raspberry pi, but then we would be able to work to add proper support for beaglebone. 

For that we would need `System.getProperty("os.arch")` from beaglebone to be able to detect it. 

Sadly, I do not have this board to verify this claim.

pschichtel, 2020-12-03:
If I knew what I can check for in os.arch, this would probably be easy to solve. Definitely sounds like a plausible explanation.

In the context of #23 I thought about adding an alternative way to load the native library instead of using the architecture detection and unpacking it to /tmp. That would allow to separately build another architecture and load it for testing here.

pschichtel, 2020-12-03:
I added a way to load a libraries from a path specified by a property.

You can build the libraries for any dockcross supported architecture like this (assuming you start in the repo root):

```bash
cd core
BUILD_ARCH=armv7a ../compile-native.sh "$JAVA_HOME" javacan-core ..
cd ../epoll
BUILD_ARCH=armv7a ../compile-native.sh "$JAVA_HOME" javacan-epoll ..
```

After that, you'll have the two `.so` files in the repo root.

Then run your program with the following options: `-Djavacan.native.javacan-core.path=path/to/libjavacan-core-armv7a.so -Djavacan.native.javacan-epoll.path=path/to/libjavacan-epoll-armv7a.so`

Alternatively you can also build JavaCAN with a single architecture by providing it to the maven build like so:

```bash
mvn clean install -Djavacan.build.arch=armv7a
```

(if you are not building on the armv7a device, you'll probably need to skip tests using `-DskipTests`)

Then you have to run your program with the options `-Djavacan.native.javacan-core.classpath=/native/libjavacan-core-armv7a.so -Djavacan.native.javacan-core.classpath=/native/libjavacan-epoll-armv7a.so`.

(note, that these properties end with `.classpath` insead of `.path`)

psobiech, 2020-12-04:
As far as I googled, BeagleBone might not report anything unique in `os.arch`. 

@pschichtel Maybe it would be better to have a single option to specify the cpu architecture eg. `javacan.cpu.arch`, that would disable auto-detection and just load `libjavacan-{core/epoll}-{javacan.cpu.arch}.so`. 

Maybe to satisfy #23 and this use case, we could first try to load system library and then fallback to loading the local one?
 
```
  try {
    	System.loadLibrary("javacan-{core/epoll}-{javacan.cpu.arch}");
  } catch (UnsatisfiedLinkError e) {
      // fallback to load the local one?
      System.loadLibrary("/native/libjavacan-{core/epoll}-{javacan.cpu.arch}.so");
  }
```

That would utilize `java.library.path` and `LD_LIBRARY_PATH`, so one would be able to install the .so in the operating system specific way, like /usr/local/lib/. 

What do you think?

pschichtel, 2020-12-08:
good point about the library search paths... probably should do that. But I still want to provide a way to explicitly specify a file

keoir, 2020-12-09:
Full Dump:

`#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xaa42e1c8, pid=10111, tid=0xb530e460
#
# JRE version: OpenJDK Runtime Environment (8.0_272-b10) (build 1.8.0_272-8u272-b10-1-b10)
# Java VM: OpenJDK Client VM (25.272-b10 mixed mode linux-aarch32 )
# Problematic frame:
# C  [libJavaCAN.so+0x21c8]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0xb5107800):  JavaThread "main" [_thread_in_native, id=10112, stack(0xb52bf000,0xb530f000)]

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0xaa42e1c8

Registers:
R0=0xaa4410d0
R1=0x00000000
R2=0x00000002
R3=0xaa441000
R4=0xaa440f18
R5=0xaa42e274
R6=0x00000004
R7=0xbebbf144
R8=0xbebbf158
R9=0xaa440f18
R10=0x00000003
R11=0xb530c6fc
R12=0x00000006
R13=0xb530c2d0
R14=0xb6f8ff41
R15=0xaa42e1c8

Top of Stack: (sp=0xb530c2d0)
0xb530c2d0:   bebbf158 ffffffff a9d0e6f8 00000004
0xb530c2e0:   bebbf144 bebbf158 00000000 b6f8ffe3
0xb530c2f0:   00000000 00000058 b530e920 b530c4e0
0xb530c300:   a9d0e858 b6fad880 00000003 b6f3242f
0xb530c310:   00000000 b530c548 b6f922c1 b530c4f0
0xb530c320:   b6f84304 b530c348 b530c3cc b6fad880
0xb530c330:   b6fae000 aaaaaaab aad1ba0c 00000003
0xb530c340:   b6fad880 b6fad880 00000000 00000000 

Instructions: (pc=0xaa42e1c8)
0xaa42e1a8:   02 20 a0 e3 4c 10 9f e5 00 00 8f e0 01 10 8f e0
0xaa42e1b8:   44 30 9f e5 01 10 60 e0 41 11 a0 e1 03 30 8f e0
0xaa42e1c8:   11 f2 11 e7 00 00 51 e3 1e ff 2f 01 2c 20 9f e5
0xaa42e1d8:   08 d0 4d e2 02 30 93 e7 04 30 8d e5 00 00 53 e3 

Register to memory mapping:

R0=0xaa4410d0: __bss_start__+0 in /tmp/JavaCAN-armv7-7504299648645545403/libJavaCAN.so at 0xaa42c000
R1=0x00000000 is an unknown value
R2=0x00000002 is an unknown value
R3=0xaa441000: <offset 0x15000> in /tmp/JavaCAN-armv7-7504299648645545403/libJavaCAN.so at 0xaa42c000
R4=0xaa440f18: <offset 0x14f18> in /tmp/JavaCAN-armv7-7504299648645545403/libJavaCAN.so at 0xaa42c000
R5=0xaa42e274: <offset 0x2274> in /tmp/JavaCAN-armv7-7504299648645545403/libJavaCAN.so at 0xaa42c000
R6=0x00000004 is an unknown value
R7=0xbebbf144 is an unknown value
[... 509 lines cut ...]

/proc/cpuinfo:
processor	: 0
model name	: ARMv7 Processor rev 2 (v7l)
BogoMIPS	: 995.32
Features	: half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc08
CPU revision	: 2

Hardware	: Generic AM33XX (Flattened Device Tree)
Revision	: 0000
Serial		: e4:15:f6:e0:a4:dc

Memory: 4k page, physical 495036k(125660k free), swap 0k(0k free)

vm_info: OpenJDK Client VM (25.272-b10) for linux-aarch32 JRE (1.8.0_272-8u272-b10-1-b10), built on Oct 26 2020 20:29:16 by "buildd" with gcc 9.3.0

time: Wed Dec  9 20:18:11 2020
timezone: UTC
elapsed time: 8.369769 seconds (0d 0h 0m 8s)

`

pschichtel, 2020-12-10:
@keoir thainks for the update. This really seems to be an incompatibility with CPU as @psobiech suggested. Can you try what I suggested [here](https://github.com/pschichtel/JavaCAN/issues/24#issuecomment-738459000) using latest master?

pschichtel, 2020-12-13:
@keoir I added armv7a to the default build and added the property `javacan.native.arch` (`-Djavacan.native.arch=armv7a`), so you can select the library

keoir, 2020-12-15:
@pschichtel Any way to get this added to the maven repo? Trying to build it locally but I don't have docker on this system.

keoir, 2020-12-15:
Got it to build and compile, got a new JNI error:
`#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb56f9bd2, pid=2591, tid=0xb52a2460
#
# JRE version: OpenJDK Runtime Environment (8.0_272-b10) (build 1.8.0_272-8u272-b10-1-b10)
# Java VM: OpenJDK Client VM (25.272-b10 mixed mode linux-aarch32 )
# Problematic frame:
# V  [libjvm.so+0x28abd2]
#
# Core dump written. Default location: /usr/local/nebula/core or core.2591
#
# An error report file with more information is saved as:
# /usr/local/nebula/hs_err_pid2591.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
`

keoir, 2020-12-15:
Got it working, but not getting this error when init is called:

`
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb56f9bd2, pid=2591, tid=0xb52a2460
#
# JRE version: OpenJDK Runtime Environment (8.0_272-b10) (build 1.8.0_272-8u272-b10-1-b10)
# Java VM: OpenJDK Client VM (25.272-b10 mixed mode linux-aarch32 )
# Problematic frame:
# V  [libjvm.so+0x28abd2]
#
# Core dump written. Default location: /usr/local/nebula/core or core.2591
#
# An error report file with more information is saved as:
# /usr/local/nebula/hs_err_pid2591.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

`

keoir, 2020-12-15:
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb56f9bd2, pid=2591, tid=0xb52a2460
#
# JRE version: OpenJDK Runtime Environment (8.0_272-b10) (build 1.8.0_272-8u272-b10-1-b10)
# Java VM: OpenJDK Client VM (25.272-b10 mixed mode linux-aarch32 )
# Problematic frame:
# V  [libjvm.so+0x28abd2]
#
# Core dump written. Default location: /usr/local/nebula/core or core.2591
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread (0xb5107800):  JavaThread "main" [_thread_in_vm, id=2592, stack(0xb5253000,0xb52a3000)]

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x00000000

Registers:
R0=0x0000deab
R1=0x00000000
R2=0x00000000
R3=0x00000000
R4=0xb5107800
R5=0x00000000
R6=0xb5107994
R7=0xb52a1910
R8=0xb596a000
R9=0x00000004
R10=0xb6d8c784
R11=0x00000000
R12=0xb51079d0
R13=0xb52a1910
R14=0xaa414374
R15=0xb56f9bd2

Top of Stack: (sp=0xb52a1910)
0xb52a1910:   b52a1918 b6d90608 b5107800 00000000
0xb52a1920:   b596a000 1af48700 b5107800 b52a1ca8
0xb52a1930:   b6f047c0 b5107800 a9cb32b8 1af48700
0xb52a1940:   b52a1958 b5107994 00000000 b313f124
0xb52a1950:   a9cb34a8 b5107800 00000004 a9cb32b8
0xb52a1960:   b52a1994 aa414374 b5107800 1af48700
0xb52a1970:   b52a19d4 b52a1998 00000000 b313f124
0xb52a1980:   a9cb34a8 b5107800 00000004 a9cb32b8

Instructions: (pc=0xb56f9bd2)
0xb56f9bb2:   71 d1 63 68 06 22 bc 60 c4 f8 d0 21 00 22 fa 60
0xb56f9bc2:   1b b1 07 f1 08 00 1f f1 6c fc 00 bf eb 07 58 bf
0xb56f9bd2:   d5 f8 00 90 79 d4 45 4a 58 f8 02 20 12 68 59 f8
0xb56f9be2:   02 20 ea b1 48 46 07 f1 18 06 b0 f7 72 fc 00 22

Register to memory mapping:

R0=0x0000deab is an unknown value
R1=0x00000000 is an unknown value
R2=0x00000000 is an unknown value
R3=0x00000000 is an unknown value
R4=0xb5107800 is a thread
R5=0x00000000 is an unknown value
R6=0xb5107994 is an unknown value
R7=0xb52a1910 is pointing into the stack for thread: 0xb5107800
R8=0xb596a000: <offset 0x4fb000> in /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/client/libjvm.so at 0xb546f000
R9=0x00000004 is an unknown value
R10=0xb6d8c784: <offset 0x191d784> in /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/client/libjvm.so at 0xb546f000
[... 497 lines cut ...]

/proc/cpuinfo:
processor       : 0
model name      : ARMv7 Processor rev 2 (v7l)
BogoMIPS        : 995.32
Features        : half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x3
CPU part        : 0xc08
CPU revision    : 2

Hardware        : Generic AM33XX (Flattened Device Tree)
Revision        : 0000
Serial          : e4:15:f6:e0:a4:dc

Memory: 4k page, physical 495036k(313680k free), swap 0k(0k free)

vm_info: OpenJDK Client VM (25.272-b10) for linux-aarch32 JRE (1.8.0_272-8u272-b10-1-b10), built on Oct 26 2020 20:29:16 by "buildd" with gcc 9.3.0

time: Tue Dec 15 22:53:20 2020
timezone: UTC
elapsed time: 8.757771 seconds (0d 0h 0m 8s)

```

pschichtel, 2020-12-15:
I deployed a snapshot to the sonatype snapshot repo including a few more null checks

pschichtel, 2020-12-16:
and one more built with debug info. I tried to poke around a bit in that function but could replicate the exact same stack trace, yet.

The build: https://oss.sonatype.org/service/local/repositories/snapshots/content/tel/schich/javacan-core/3.0.0-SNAPSHOT/javacan-core-3.0.0-20201216.014355-3.jar

pschichtel, 2021-02-25:
@keoir Would it be possible to provide the outputs of `cat /proc/cpuinfo`, `uname -p` and `lscpu`? I want to improve the architecture detection for the multi architecture version (#31). The output of a java program calling `System.out.println(System.getProperty("os.arch"))` would also be awesome.

pschichtel, 2021-03-08:
I'm closing this issue since there hasn't been any more feedback with all the changes around library loading since then, especially #31.

closed 2021-03-08 by pschichtel