pschichtel/JavaCAN #30: Raspberry Pi Zero support

Szbuli, 2021-02-21:
I am having similar problems as in #24.

Should Raspberry Pi 3/4/zero etc work out of the box with the library? 

If I can help to resolve this problem let me know.

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xa8d32ce0, pid=22763, tid=0xa775e460
#
# JRE version: OpenJDK Runtime Environment (8.0_212-b01) (build 1.8.0_212-8u212-b01-1+rpi1-b01)
# Java VM: OpenJDK Client VM (25.212-b01 mixed mode linux-aarch32 )
# Problematic frame:
# C  [libjavacan-core.so+0x1ce0]
```
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xa8d32ce0, pid=22763, tid=0xa775e460
#
# JRE version: OpenJDK Runtime Environment (8.0_212-b01) (build 1.8.0_212-8u212-b01-1+rpi1-b01)
# Java VM: OpenJDK Client VM (25.212-b01 mixed mode linux-aarch32 )
# Problematic frame:
# C  [libjavacan-core.so+0x1ce0]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0xa7626c00):  JavaThread "CAN_RECIEVE_THREAD" [_thread_in_native, id=22778, stack(0xa770f000,0xa775f000)]

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0xa8d32ce0

Registers:
R0=0xa8d460c0
R1=0x00000000
R2=0x00000002
R3=0xa8d46000
R4=0xa8d45f18
R5=0xa8d32d8c
R6=0x00000009
R7=0xbee385e4
R8=0xbee3860c
R9=0xa8d45f18
R10=0xb6fb8970
R11=0xa775c52c
R12=0x0000002f
R13=0xa775c498
R14=0xb6f982e8
R15=0xa8d32ce0

Top of Stack: (sp=0xa775c498)
0xa775c498:   00000000 01a11cb8 00000009 bee385e4
0xa775c4a8:   bee3860c 00000000 b6fb8970 b6f983ec
0xa775c4b8:   ffffffff 01a11cb8 00000003 00000003
0xa775c4c8:   00000008 b6fb9578 b6fb8970 b6f9c394
0xa775c4d8:   a775c52c b6f9c160 01a11cb8 00000000
0xa775c4e8:   00000000 439b7805 9ab782de 77e0ac4b
0xa775c4f8:   a775c4e0 00000000 a775c700 80000001
0xa775c508:   c3c66c0f 00000000 a775c6f4 80000001 

[... 536 lines cut ...]
/proc/cpuinfo:
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 997.08
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000c380ffda
Model		: Raspberry Pi Zero W Rev 1.1

Memory: 4k page, physical 442224k(132160k free), swap 1048572k(1047292k free)

vm_info: OpenJDK Client VM (25.212-b01) for linux-aarch32 JRE (1.8.0_212-8u212-b01-1+rpi1-b01), built on Mar 29 2019 13:54:02 by "root" with gcc 8.3.0

time: Sun Feb 21 18:44:40 2021
timezone: GMT
elapsed time: 17 seconds (0d 0h 0m 17s)

```

pschichtel, 2021-02-25:
@Szbuli It seems the PI Zero useses armv6, but JavaCAN is currently only build for armv7(a). Can you try what I sad in #24 for building a "single architecture version" for armv6? if that works, I'll add it to the default pipeline and try to adjust the architecture detection.

If you have trouble, I could build a version for you, but I don't have the hardware to test it myself.

Szbuli, 2021-02-25:
Thanks, I will try to build it and test it and get back to you soon.

pschichtel, 2021-02-25:
Would it be possible to provide the outputs of `cat /proc/cpuinfo`, `uname -p` and `lscpu`? I want to improve the architecture detection for the multi architecture version (#31). The of a java program calling `System.out.println(System.getProperty("os.arch"))` would be awesome.

Szbuli, 2021-02-25:
Sure.

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000c380ffda
Model           : Raspberry Pi Zero W Rev 1.1

pi@raspberrypi:~ $ sudo uname -p
unknown

pi@raspberrypi:~ $ lscpu
Architecture:        armv6l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               7
Model name:          ARM1176
Stepping:            r0p7
CPU max MHz:         1000.0000
CPU min MHz:         700.0000
BogoMIPS:            697.95
Flags:               half thumb fastmult vfp edsp java tls

System.out.println(System.getProperty("os.arch"));
arm

pschichtel, 2021-02-25:
Thanks!

> System.out.println(System.getProperty("os.arch"));
> arm

That's exactly why the architecture detection needs improving before I can automatically load libraries for the correct arm architecture

pschichtel, 2021-03-08:
So I refactored the build setup in a way, that completely drops runtime architecture detection. Instead I've put the native libs in dedicated jars per architecture with a classifier on them. This version is currently the latest deployed build in sonatype-snapshots. You just add a dependency to tel.schich:javacan-core:3.0.0-SNAPSHOT:armv6 (same for the epoll module) in addition to the dependency to the default jar.

pschichtel, 2021-03-15:
Did you have a chance to test this?

Szbuli, 2021-03-15:
Thanks, it works like a charm!

Is it still possible to support multiple architectures with the same jar? It would be nice if I did not have to manage seperate depencenies. I will have the same app running on zero, rpi3, rpi4, etc..

I am not sure what happens now if I have multiple architecture depencies, eg. armv6, armv7, etc.

pschichtel, 2021-03-16:
Unless you have some other means for detecting/selection the arch, I would recommend against arch detection. The native-code jars all have the library at `/native/javacan-<module>.so`, so shading in multiple will run into conflicts.

Your options:
1. You could bundle all the jars for the different archs and dynamically load the right one at runtime.
2. You could try to relocate the *.so files to different locations and at runtime explicitly set the `javacan.native.javacan-<module>.classpath` property (maven-assembly-plugin might help here).
3. You could also bundle just the *.so files with your main application and set the `javacan.native.javacan-<module>.path` property to load them from the filesystem.
4. In case you are building native OS packages at some point, you could install the appropriate *.so files into the systems `LD_LIBRARY_PATH`.

Does any of these options work for you?

pschichtel, 2021-03-22:
I'll be releasing JavaCAN 3.0 today unless there are objections against the currently chose approach.

Szbuli, 2021-03-23:
Thanks for your help! One of your suggestion will definitely work!

closed 2021-03-23 by Szbuli