pschichtel/JavaCAN #32: Examples or API documentation

DoctorZIP, 2021-03-23:
Hi,
First of all, I want to thank you for a great job. It's really easy to kickstart and check if the system works in a first approach but, it is a lack of examples of descriptions on more complex things like filters and non-blocking API. Could you please provide such information or point out where it could be found?
Thank you!

pschichtel, 2021-03-26:
The unit tests provide some usage examples for various things. The general API design of the core module is very close to any other Channel implementation and epoll module is very close to Java's Selector API (actually, in 2.x it implements the Selector API, but that implementation is not completely correct). Options are named exactly like they are in SocketCAN. I intentionally tried to stay as close as possible to existing and familiar APIs, so users can transfer existing knowledge from elsewhere and I can avoid writing lots of documentation :smile: . I do see your point though.

If you have specific things you need, please mention them and I'll see if I can write up some documentation and/or an additional test. Do understand though, they I'll only be writing documentation for version 3.0 and not for the 2.x branch.

DoctorZIP, 2021-03-26:
Okay. Thank you! When do you plan to release version 3?

closed 2021-03-26 by DoctorZIP

pschichtel, 2021-03-26:
Actually I just did. For some reason the deployment to maven central did not work, looking into that later today...

pschichtel, 2021-03-27:
@DoctorZIP fixed it 3.0.1 is now available from maven central

DoctorZIP, 2021-03-27:
> @DoctorZIP fixed it 3.0.1 is now available from maven central

Looks like no JARs are included https://repo.maven.apache.org/maven2/tel/schich/javacan/3.0.1/

pschichtel, 2021-03-27:
the javacan artifact itself is now just an aggregator. javacan-core and javacan-epoll are what you need now:

* https://repo.maven.apache.org/maven2/tel/schich/javacan-core/3.0.1/
* https://repo.maven.apache.org/maven2/tel/schich/javacan-epoll/3.0.1/