pschichtel/JavaCAN #38: Project Panama

pschichtel, 2022-03-26:
It would be nice to have a version of JavaCAN that uses Project Panama for SocketCAN and epoll interaction instead of JNI, making this a pure Java project.

jextract might come in handy for generating the low-level glue-code:
http://cr.openjdk.java.net/~mcimadamore/panama/jextract_distilled.html

splatch, 2022-05-12:
BTW have you thought of giving a try to jnr-ffi or JNA? I think both allow to cut JNI part with some performance sacrifices. I saw it used with dbus (for unix sockets) and systemd (to call its library). Both work with present releases of JVM.

pschichtel, 2022-05-12:
Yes, both would work but face the exact same challenge as project panama.

pschichtel, 2024-02-28:
labeling this for the next major version. I don't think it makes sense to do this without rethinking some API decisions.

pschichtel, 2024-03-19:
Java 22 makes this Project Panama stable: https://openjdk.org/projects/jdk/22/

pschichtel, 2025-07-16:
Java 25 (first LTS to include panama) will be released this september, that might be time to revisit this.