pschichtel/JavaCAN #42: Expand library for more transports

splatch, 2022-05-12:
Recently I had a look on CAN-over-ethernet as I was forced to do a remote debug of some CAN communication.
I realized that it is not that hard to make basic client of `socketcand` as I was able to do that using telnet client. This made me curious if we could get other drivers which would work under Windows and not require native libraries. I started to look at other known CAN drivers/transports/interfaces out there such as `slcan` and `gs_usb`. I think these are not that far away and could be implemented. Effectively it would open JavaCAN to wider pool of users who are stuck to Windows.

I am not certain if it would fit into idea of JavaCAN as for now this project is strictly speaking focused on socketcan bridge. It will being also more dependencies (ie. serial port handling).
I wouldn't want to implement that in Apache PLC4X cause its internal transport mechanism is bound to netty which complicates a lot basic protocol work.

What your take on that?

splatch, 2022-05-12:
Hey Phillip,
Thanks for fast answer. I am aware that bounding between socketcan and JavaCAN is present. I don't really mind it cause it gave this project a structured approach. The `CanFrame` is good enough for beginning, requires minor adjustments I believe. BCM and ISO-TP support you made is also an interesting area which appears in `socketcand`.

I think that first one I would try is socketcand as it is just a TCP read/write interface for socketcand. It can be done with fairly basic socket client, without extra dependencies. It also solves my immediate need to have a remote access to can interface with diagnostics software. I would put other implementation for later, if we get over socketcand. I have few [ucan devices](https://github.com/UsbCANConverter-UCCbasic/UCCB_GS_Embedded) dongles (yet another fork of candleLight), their firmware can be swapped to `slcan` and cross tested under windows.

One thing I noted while playing with `socketcand` is the timing information which is provided as part of frame which is currently not supported by API (see #5):
```
rcv: < hi >
snd: < open can0 >
rcv: < ok >
snd: < rawmode >
rcv: < ok >
rcv: < frame 738 1652295808.477464 05 >
rcv: < frame 738 1652295808.479889 058A010000000000 >
rcv: < frame 728 1652295809.835495 05 >
```
(receive frame format is `< COB sec.usec data >`, not sure about sending as I didn't go that far.

pschichtel, 2022-12-16:
I think a better approach for this would be a layer on top of JavaCAN

pschichtel, 2024-02-28:
_if_ I tackle this, it will come with breaking changes to the API, so javacan 4.