pschichtel/JavaCAN #48: examples of 'compile-native.sh' invocation? maybe github action?

rusefillc, 2023-11-19:
core artifacts are available on maven

I am a bit confused about the source of native binaries. Are those available anywhere or is everyone compiling for themselves? I see compile-native.sh script and some arguments are way over my head. Any chance to add at least one example of compile-native.sh invocation for some most default target platform like x86 ubuntu or debian?

pschichtel, 2023-11-19:
Hi, `compile-native.s`h is not supposed to be invoked directly, it's invoked by maven to run the cross compilation with dockcross. If you want to build custom architecture you can invoke maven with appropriate options according to https://github.com/pschichtel/JavaCAN#building.

The native components for all pre-built architectures (which includes x86_64) are published as separate jars each with their own classifier:

* https://repo1.maven.org/maven2/tel/schich/javacan-core/3.2.4/
* https://repo1.maven.org/maven2/tel/schich/javacan-epoll/3.2.4/

So in your https://github.com/rusefi/rusefi/commit/fea753f5808b284e1d9f8ec32280a7a135a5b72f change I see you probably only need this:

```gradle
api group: 'tel.schich', name: 'javacan-core', version: '3.2.0'
implementation group: 'tel.schich', name: 'javacan-core', version: '3.2.0', classifier: 'x86_64'
```

Does that help?

rusefillc, 2023-11-19:
Omg thank you for the extremely detailed response!

pschichtel, 2023-11-19:
@rusefillc I made the README a little more explicit about where to find the artifacts.

commit 0140dbcf (2023-11-19): be more explicit about where to find native libs

closes #48
commit b6f72565 (2023-11-19): be more explicit about where to find native libs

closes #48
closed 2023-11-19 by pschichtel