pschichtel/JavaCAN #49: github actions

rusefillc, 2023-11-19:
github actions are amazing for so many reasons
* green checkbox instills confidence
* github actions could produce snapshot artifacts for each commit
* github scripts are self-documenting build procedure

pschichtel, 2023-11-19:
There is an action, but it fails and I haven't looked at why, probably something related to dockcross. In its current state JavaCAN doesn't need a lot of changes beyond documentation, so a workflow brings little value, that's why I haven't bothered yet.

pschichtel, 2023-11-19:
nvm I guess: the build does work

https://github.com/pschichtel/JavaCAN/actions/runs/6922072977

it does built the project and publish the snapshots, but it doesn't run tests because I need a vcan device and the isotp module for that. I haven't looked into whether this is possible.

Do you have anything specific in mind?

pschichtel, 2023-11-19:
Just tried it:

```
Run ./setup-vcan.sh
  ./setup-vcan.sh
  shell: /usr/bin/bash -e {0}
  env:
    JAVA_HOME: /opt/hostedtoolcache/Java_Temurin-Hotspot_jdk/8.0.382-5/x64
    JAVA_HOME_8_X64: /opt/hostedtoolcache/Java_Temurin-Hotspot_jdk/8.0.382-5/x64
Error: Unknown device type.
Cannot find device "vcan0"
modprobe: FATAL: Module can_isotp not found in directory /lib/modules/6.2.0-1015-azure
Failed to load the ISOTP module, related unit tests will probably fail
```

The testing story pretty much ends here unless someone sponsors a custom actions runner with vcan, can_raw, can_bcp and can_isotp supported

pschichtel, 2023-11-19:
https://github.com/pschichtel/JavaCAN/actions/runs/6922241669/job/18828737259

closed 2023-11-19 by pschichtel