pschichtel/JavaCAN #54: J1939 Support

pschichtel, 2024-01-02:
As "requested" by @JohnLussmyer one of the missing J1939 protocol should be added. I have little to no knowledge about the protocol, so any reference code would be helpful to implement it. I assume it would be similar to the RAW channel.

JohnLussmyer, 2024-01-14:
I've been looking into making some changes so I could test things.
So far, I can't build due to the java 8 requirement.  openjdk-8-jdk doesn't seem to exist for a Raspberry Pi 4.
Do you think it will work with Java 17 if I remove the check for java 8?

pschichtel, 2024-01-14:
> I've been looking into making some changes so I could test things.
Awesome!

>  openjdk-8-jdk doesn't seem to exist for a Raspberry Pi 4.
OpenJDK 8 for aarch64: https://adoptium.net/temurin/releases/?os=linux&arch=aarch64&package=jdk&version=8

> Do you think it will work with Java 17 if I remove the check for java 8?
Compilation will most certainly not work in anything >8 because they changed some paths around and I don't think the current build scripts support the newer layout. JavaCAN can run on any JRE >=8 though, it's compiled with jdk 8 to ensure that.

The build infrastructure is centered around cross-building other architectures from x86_64 using dockcross, currently there isn't any way to build JavaCAN with the "host architecture". Dockcross is also only available for x86_64 if I'm not mistaken.

So my recommendation is: Build JavaCAN on an x86_64 machine and transfer the jar over to your raspberry pi.

Is that viable for you?

JohnLussmyer, 2024-01-15:
Ok, I finally got a x64 linux box up and running.  Everything is installed. 
Note that I carefully did NOT change any of your code.
When I ran the full build, I get an error (below).  Any ideas?

[INFO] Running tel.schich.javacan.test.util.IsotpListenerTest
[ERROR] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.058 s <<< FAILURE! -- in tel.schich.javacan.test.util.IsotpListenerTest
[ERROR] tel.schich.javacan.test.util.IsotpListenerTest.testBroker -- Time elapsed: 0.054 s <<< ERROR!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to write to the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'
        at tel.schich.javacan.SocketCAN.write(Native Method)
        at tel.schich.javacan.AbstractCanChannel.writeSocket(AbstractCanChannel.java:237)
        at tel.schich.javacan.IsotpCanChannelImpl.write(IsotpCanChannelImpl.java:102)
        at tel.schich.javacan.test.util.IsotpListenerTest.testBroker(IsotpListenerTest.java:85)

pschichtel, 2024-01-15:
Seems like something is off about the ISOTP socket `a` in the test.

Can you confirm that this works:

1. Open 1. shell and run `isotprecv -s 7E0 -d 7E8 vcan0`
2. Open 2. shell and run `isotpsend -s 7E8 -d 7E0 vcan0 <<< "a"`
3. 1. shell should print `0A`

Also please share the output of `lsmod | grep can`.

JohnLussmyer, 2024-01-15:
Shell 1 echoed just fine.

john@Linux-HP:~$ isotprecv -s 7E0 -d 7E8 vcan0
0A
john@Linux-HP:~$ lsmod | grep can.
vcan                   12288  0
can_dev                57344  1 vcan
can_raw                16384  0
can_isotp              28672  0
can_bcm                24576  0
can                    24576  3 can_isotp,can_raw,can_bcm
john@Linux-HP:~$

JohnLussmyer, 2024-01-15:
I just re-read your README.md - and realized I have no idea what to do for this line of instruction:

•The can-isotp kernel module loaded (Kernel 5.10 with CONFIG_CAN_ISOTP enabled or the [out-of-tree module](https://github.com/hartkopp/can-isotp))

pschichtel, 2024-01-15:
you have the can_isotp module loaded as shown by your lsmod output, that's why I asked for it.

Which kernel version are you running? (`uname -a`)

Also:

can you try running the test with the following added after line 69 in file `tel/schich/javacan/test/util/IsotpListenerTest.java`:

```java
a.configureBlocking(true);
b.configureBlocking(true);
```

JohnLussmyer, 2024-01-15:
john@Linux-HP:~$ uname -a
Linux Linux-HP 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

[INFO] Running tel.schich.javacan.test.util.IsotpListenerTest
[ERROR] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.058 s <<< FAILURE! -- in tel.schich.javacan.test.util.IsotpListenerTest
[ERROR] tel.schich.javacan.test.util.IsotpListenerTest.testBroker -- Time elapsed: 0.054 s <<< ERROR!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to write to the socket - errorNumber=11, errorMessage='Resource temporarily unavailable'
        at tel.schich.javacan.SocketCAN.write(Native Method)
        at tel.schich.javacan.AbstractCanChannel.writeSocket(AbstractCanChannel.java:237)
        at tel.schich.javacan.IsotpCanChannelImpl.write(IsotpCanChannelImpl.java:102)
        at tel.schich.javacan.test.util.IsotpListenerTest.testBroker(IsotpListenerTest.java:87)

pschichtel, 2024-01-15:
Ok, do other tests work? E.g. Just delete this one and run the build again

JohnLussmyer, 2024-01-15:
Deleted isotplistenertest.java - and the build ran with no errors.

JohnLussmyer, 2024-01-16:
Now I get to start writing tests to see if my new code works at all....

JohnLussmyer, 2024-02-10:
Question: I just noticed that the Pi 4 uses "Broadcom BCM2711, Quad core Cortex-A72 (ARM v8) 64-bit SoC @ 1.8GHz"
Any chance that the code supports the ARM v8 processor?

pschichtel, 2024-02-10:
Isn't still just aarch64? I'd assume the aarch64 build should just work, but I have no hardware to actually test it. Is this just a question or did you actually observe an issue? Architecture auto-detection in Java is difficult on all the different arm variants, so if you use that it might have picked the wrong library.

JohnLussmyer, 2024-02-10:
Just a question, since I saw that ArmV6 and V7 were explicit native code supported.

JohnLussmyer, 2024-02-10:
And I just tried building the "all-architectures".  It got up to:
[INFO] --- exec-maven-plugin:3.1.1:exec (compile-native-riscv32) @ javacan-core ---
Compiling for: riscv32 (dockcross: docker.io/dockcross/linux-riscv32:20231110-9476e91)
....
Copying blob sha256:28469740a38f7f50a08906c874b71b52323ecbab72b05ab940a5891016b73ec9
Error: writing blob: storing blob to file "/var/tmp/storage3447402512/13": happened during read: unexpected EOF
[ERROR] Command execution failed.
org.apache.commons.exec.ExecuteException: Process exited with an error: 125 (Exit value: 125)

I have plenty of disk space.

pschichtel, 2024-02-10:
are you using podman? it's default storage configuration might be "sub optimal" on some distros. The script prioritizes podman if available, but you can force it to use docker using the ENV var I just added on master: `RUN_CONTAINER_COMMAND="docker run --rm" mvn ...`

pschichtel, 2024-02-10:
ok, I see that github actions are also failing with that error using podman, maybe there is something up with the dockcross images or a new podman version...

JohnLussmyer, 2024-02-10:
Ahh, ok.  I'll ignore it for now, and update/test later.  I'm still working on adding J1939 changes. (yeah, I'm slow since I am NOT familiar with CAN, J1939, or Linux much.)

pschichtel, 2024-02-10:
If you are comfortable sharing your draft state early I might be able to help, but no pressure :) Either way I'm happy to help/assist.

pschichtel, 2024-02-10:
Ok CI works again with the latest dockcross version, it also builds locally with podman. Not sure what exactly broke there...

JohnLussmyer, 2024-02-10:
I need to first get the general structure done.  I've been mostly busy with other things the past few weeks.  (WAY too much to do...)

And frankly, I'm not sure HOW share a draft.  I've never needed to work with Github before.

JohnLussmyer, 2024-02-11:
Okay, I've got all the basics done, just haven't gotten any tests that actually USE the can channel to work.
This is happening on the write() call, the create seems to have worked.
I can see that there are some can protocol options on the modprobe line in the setup-vcan.sh, but I've been unable to find ANY docs on what those do.  I did try adding a can_j1939 just for the heck of it, but that didn't help.

[ERROR] tel.schich.javacan.test.J1939CanSocketTest.testLoopback -- Time elapsed: 0.009 s <<< ERROR!
tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to write to the socket - errorNumber=13, errorMessage='Permission denied'

time to have dinner...

pschichtel, 2024-02-11:
> And frankly, I'm not sure HOW share a draft. I've never needed to work with Github before.

1. fork this repo on github
2. set it's git-url as your origin (`git remote set-url origin <url of you fork>`)
3. add this repo as a second remote (`git remote add upstream https://github.com/pschichtel/JavaCAN.git`)
4. create a new branch (if you haven't already) like `feature/j1939`
5. push the branch with your changes to your fork
6. go your fork on github, it should show you a button to create a pull request from the newly pushed branch
7. if you want to rebase onto my master branch just `git fetch upstream && git rebase upstream/master`

> can see that there are some can protocol options on the modprobe line in the setup-vcan.sh, but I've been unable to find ANY docs on what those do.

those are not protocol options, those are just kernel module names. So adding `can_j1939` is exactly right.

> [ERROR] tel.schich.javacan.test.J1939CanSocketTest.testLoopback -- Time elapsed: 0.009 s <<< ERROR!
> tel.schich.javacan.platform.linux.LinuxNativeOperationException: Unable to write to the socket - errorNumber=13, errorMessage='Permission denied'

interesting, so you apparently allowed to create the socket, but not allowed to write to it. that seems odd, it would be create to be able to test/debug this locally.

JohnLussmyer, 2024-02-11:
I'll see if I can do the github process.

JohnLussmyer, 2024-02-11:
Well... I MAY have done it?  Not sure about dealing with the Pull Request on github, as their instructions don't match the screens I have.

pschichtel, 2024-02-11:
yes, looks good. thanks, I'll give that try

pschichtel, 2024-03-09:
I consider this done. If you encounter missing functionality or problems please report dedicated issues.

closed 2024-03-09 by pschichtel

splatch, 2024-03-11:
While software simulator is hard to find I began looking for FD enabled DIO and found this: https://www.microcontrol.net/en/portfolio/i-o-modules/snap/mcan-8-dio-snap/, turns out it can also work in J1939 mode.

JohnLussmyer, 2024-03-11:
Anyone have documentation on the Canoe "CAN bus log file in asc or blf format"?
I need to ask a question of the manufacturer of the device I'm working with - and they won't answer ANY questions without a log file in one of those formats.
I'm not going to buy Canoe to generate a log file, but I am perfectly capable of writing a program to create one if I know what the format is.

splatch, 2024-03-11:
Maybe you could look at [Kayak](https://github.com/dschanoeh/Kayak), its archived but it contains support for some of common CAN log formats.

pschichtel, 2024-03-12:
The CANoe stuff seems very proprietary. Unless there is some open spec or a trail version of the software, I don't see how we'll be able to implement this.

JohnLussmyer, 2024-03-12:
Someone on another list just pointed me to https://savvycan.com/ which might have a file format conversion for it.

pschichtel, 2025-02-19:
@JohnLussmyer FYI the kernel has pull requests to be merged that will improve J1939 support and I think some improvements have already been merged in the meantime.